Xinyuan-LilyGO/T-Wristband #2: Sleep mode battery duration

TioRuben, 2020-01-23:
¿How much battery time do you get in sleep mode? I've tried with this code and mine [https://github.com/TioRuben/TTGO-T-Wristband-First-Steps](https://github.com/TioRuben/TTGO-T-Wristband-First-Steps) and barely get 1 day of battery in sleep mode.

PolyMeilex, 2020-01-25:
Same here, I have no idea how to improve it yet

TioRuben, 2020-01-25:
> Same here, I have no idea how to improve it yet
> 
> And btw, could you explain why did you use those values in your battery code
> 
> ```c++
> #define BATTERY_MIN_V 2.9
> #define BATTERY_MAX_V 4.0
> ```
> 
> I thought that those are from 3.27V to 4.2V, did I miss something?

@PolyMeilex You are right, the battery management in my code it's still in progress and the code is reused from another project with Li-Ion battery, I should change to the values you propose, as the T-Wristband battery is Li-Po. But also need to port the ADC calibration code from LilyGo.

About the battery life, I've improved a little bit by adding some delays after LCD switch off and adding a reset before SLPIN:
```c++
  tft.writecommand(ST7735_SWRESET);
  delay(100);
  tft.writecommand(ST7735_SLPIN);
  delay(150);
  tft.writecommand(ST7735_DISPOFF);
```
(Point 9.12 of [Manufacturer's documentation](https://www.displayfuture.com/Display/datasheet/controller/ST7735.pdf)

And also disabling RTC timers, alarm and CLK:
```c++
  rtc.clearTimer();
  rtc.disableAlarm();
  rtc.disableCLK();
  rtc.disableTimer();
```
Now I get ~3 days of battery in sleep mode.

Will update my repo during the weekend.

PolyMeilex, 2020-01-25:
~3 days?? OMG! You are my saviour, thank you very much, I will test it in on my device right away

TioRuben, 2020-02-09:
Another thing to take in account is that, according to the schematic, the backlight LED seems not to have a current limiting resistor. In the case of my unit, it heats up a little bit near the union between the right side of the TFT and touch button. Maybe the current limiting resistor is embedded in the own display, but afterwards, I've configured PWM with about 72% duty cycle in TFT_BL pin without noticeable brightness change and a noticeable heat decrease.

Here is my TFT initialization code:
```c++
  tft.init();
  tft.setRotation(1);
  tft.setSwapBytes(true);
  tft.fillScreen(TFT_BLACK);
  ledcSetup(0, 5000, 8);
  ledcAttachPin(TFT_BL, 0);
  ledcWrite(0, 185);
```

The code can be improved to provide more duty cycle at day hours than in night hours.

primaldev, 2020-03-22:
Also putting the motion sensor to sleep will help a lot. 

Just ad this to TioRuben's list:
`IMU.setSleepEnabled(true);`

hpsaturn, 2020-04-12:
@primaldev this is a good improvement, because not all of us, maybe need wake up the wristband after some  movements. Maybe it will be a setting from the user in the future for improve the battery life. I tested it on @TioRuben code and works fine.

PhilColbert, 2020-05-15:
I have the newer imu, have added all the ideas in this thread, but still getting 8-9 hours on sleep..... not much point of a watch that lasts 8 hours ?

Anyone got any further with this?

Thanks

boogie, 2020-05-16:
It's a pain for me as well, as I have bought a lot of t-wristbands. :/ Hope there's a solution for this.

PhilColbert, 2020-05-16:
Fortunately, I only got two to test, I hope theres a solution to this aswell as they look amazing.

The esp32 in deep sleep is supposed to pull between .15ma and 10µA so some other peripheral is still draining the battery.

The LSM9DS1 specs say 4.6ma during use and around 2.4 ma in sleep mode.
The pcf8563  is negligible.
The me6217 is 100uA.
So perhaps the screen is still drawing power ?

Ideas?

PhilColbert, 2020-05-18:
Just got a reply from lilygo saying the wristband should do 400UA in sleep... hopefully they will be able to get this, that would give around 8 days running which would be acceptable :)

PhilColbert, 2020-05-22:
No reply from lilygo, tried emailing them again.... not best service :(

boogie, 2020-05-28:
For a different hardware I have seen a tip, that you should set all the (SPI) ports of the display to LOW with digitalWrite, and that is the only way to reduce the power consumption. I have no tools now to check this theory for this band, but maybe you have.

PhilColbert, 2020-05-30:
Is that before sleep you mean ? I ran deep sleep at the start of setup() and managed to get 1.5ma current draw, still far away from the 400UA specified by lilygo.

I had a reply from lilygo , linking me to the documentation for esp deep sleep, completely pointless information again..

I have emailed back again asking how they get 400UA, I am feeling the lack of response isnt good

TheNitek, 2020-05-30:
Actually that may not be to bad of an advice because the ESP pins will float in deep sleep but there is an option to tell the ESP to keep certain pins in a specific state even while sleeping, so this might help to keep the display REALLY disabled while deep sleep.

TheNitek, 2020-05-30:
https://github.com/espressif/arduino-esp32/issues/2712

nav-andrea, 2020-06-08:
Hello, in our project we need to use BLE and WiFi
We've disabled all the process as you explain, also di display, and we use BLE 1 time every 30 seconds and WiFi 1 time every 15 minutes (using delay). 
From 4.3V (max load) to 3.2V we've about 1 hour of usage. As the power reach 3.2V the device turn off. How can you have 3 days in sleep mode?
During sleep mode what kind of functionality could you have?

PhilColbert, 2020-06-09:
Unfortunately still no answer from lilygo after 3 emails about this problem :(

Renders the device useless for most cases I would think :(

boogie, 2020-06-09:
Indeed, it's a crap this way. I still hope it is fixable.

boogie, 2020-06-13:
**DONT BUY LILYGO® TTGO T-Wristband DIY Programmable Smart Bracelet**
It is useless, and they are still selling it.

I wrote to their technical email address (zhang@lilygo.cc) - no response, and to their business (lily@lilygo.cc) and sales (sales@lilygo.cc) about these issues. The "Quality guarantee" link is not working on their website (http://www.lilygo.cn/). I'm trying to get my money back.

boogie, 2020-06-15:
I got an answer. It is for development, I can upgrade the battery. I think it won't solve any of these problems.

hpsaturn, 2020-06-17:
@boogie but this is the reality, it is a development wristband, maybe can you work now in a possible software with this version, in mind in the future maybe **LilyGO** will launch a device with these issues fixed. For me its fine.

PhilColbert, 2020-06-17:
Thats pretty shocking, no where on their advertising say its for development.

No use for me if it only lasts a few hours!

boogie, 2020-06-18:
@hpsaturn I'm happy if it's good for you. Not for me. I've checked their Aliexpress site again, and this device is not advertised as a development wristband at all.

Also think it's not supportive that they're not responding in this thread, and I'm looking forward to see if they can share a code that proves the 400uA consumption they told us in email.

PhilColbert, 2020-06-18:
I have sent 6 emails so far with one response which was a link to the esp_deep_sleep code.... pointless.

Managed to get the sleep down to 1.5ma , if you just go into deep sleep on boot.... but still not good enough

Bearic, 2020-06-24:
I just wanted to chime in here. I received mine a few days ago, and after figuring out how to upload code, I just turned off the factory tests. I put it in deep sleep overnight ~10 hours and it's still at 3.88 V according to the display (~70%?). Maybe there were library updates that fixed it? Though I just noticed the time is off by about 30 minutes... Not sure why. Looking forward to playing with it some more.

hpsaturn, 2020-06-25:
Maybe @Bearic you could test [this firmware](https://github.com/TioRuben/TTGO-T-Wristband) too.

Bearic, 2020-06-27:
The battery was still going after 48 hours (3.6V I think), mostly on deep sleep but with occasional turning on.

I have been using Arduino to upload sketches, once have time to figure out how to use platformio, I will give it a try and report back.

boogie, 2020-06-27:
@Bearic which hardware version you have? Maybe they have fixed the issue.

lewisxhe, 2020-06-28:
The latest current consumption has been updated to a minimum of 1.6mA. Compared to the MPU9250, it is much higher, and currently it cannot be lower. The current consumption in the product introduction is MPU9250, I have notified them to change.  https://github.com/Xinyuan-LilyGO/LilyGo-T-Wristband/blob/7a265456f2521a0beee733de135a281c8bbaa67d/examples/T-Wristband-LSM9DS1/T-Wristband-LSM9DS1.ino#L907

utya, 2020-07-06:
@lewisxhe @TioRuben  so Which version of the band is better to take? I need theband able to work for 5 hours with IMU and send data via bluetooth once a minute

lewisxhe, 2020-07-06:
Send once a minute, 5 hours? This is impossible unless the battery is replaced

utya, 2020-07-06:
sadly

andrea-nistico, 2020-07-15:
Can we mount a bigger battery on this thing?

rkilchmn, 2020-08-03:
I was just researching LilyGO-T-Wristband when I found this thread. What I was thinking is that the board should have a soft latch circuit (like https://www.instructables.com/id/Soft-Latch-Power-Switch-Ardweeny/ i have used in one of my projects) that completely cuts of the power via MOSFET and can be switched on either by signal from the tap sensor or via a programmable wake-up from the RTC, This way you can activate the device, do some stuff like show time on tap or periodically send data on RTC wake up and then send signal to power down.

DeviousPenguin, 2020-08-11:
Would getting the model with LSM9DS1 instead of MPU9250 give lower sleep mode power usage?

myleskeller, 2020-08-15:
> LilyGo-T-Wristband/examples/T-Wristband-LSM9DS1/T-Wristband-LSM9DS1.ino

so which version of the board would have the lowest power consumption?
MPU9250 or LSM9DS1?

DeviousPenguin, 2020-08-15:
For what it's worth, I've ordered a LSM9DS1 board, once it arrives I'll see how it performs.

I ordered it with a 105maH battery so if ~1mA is possible then it could be useful, we shall see... :)

wide-area-devices, 2020-08-23:
I have a number of LSM9DS1 bands, this newer (and poorer) design is total and utter rubbish, I'm wanting to use the IMU but only for temperature, once I've turned it on and then everything to turn it off again, I'm at around 4.2ma still with ESP in timer sleep......what! why they didnt power it from an I/O pin so I could chop it totally is beyond me. I'm fairly sure the MPU9250 is the better device

Yes you could add a bigger battery but it probably wont fit inside the case, maybe that 105mah will just

Tomorrow I'm hot airing the LSM off, see where that leaves me and then see if I can fixup anHTU21D somehow

wide-area-devices, 2020-08-24:
I have resorted to hot airing the LSM9DS1 off, ESP deep sleep with timer now at 0.518ma down from 1.67ma with the IMU in place, why they didnt hang it off an I/O pin is beyond me or give us a library that makes it sleep efficiently

DeviousPenguin, 2020-08-24:
@wide-area-devices thanks for sharing your findings, if I had a hot air rework setup I'd try the same thing myself.

genotix, 2020-08-25:
> I have resorted to hot airing the LSM9DS1 off, ESP deep sleep with timer now at 0.518ma down from 1.67ma with the IMU in place, why they didnt hang it off an I/O pin is beyond me or give us a library that makes it sleep efficiently

Lol I believe we have been doing the exact same thing the past 2 days.

I agreen on this and what is kindof confusing too is how TioRuben explains -HOW- you need to modify the library files since the issue is in the LSM9DS1 library defaults actually.

So here a very 4 step tutorial on what you should actually do in order to gain the control of the LSM9DS1

1. Go to the SparkFun LSM9DS1 library _source folder_ e.g. **SparkFun_LSM9DS1_Arduino_Library-master/src**
2. Edit SparkFunLSM9DS1.h and search for **initGyro()**
3. Now **CUT** this full bit:
```
	// initGyro() -- Sets up the gyroscope to begin reading.
	// This function steps through all five gyroscope control registers.
	// Upon exit, the following parameters will be set:
	//	- CTRL_REG1_G = 0x0F: Normal operation mode, all axes enabled. 
	//		95 Hz ODR, 12.5 Hz cutoff frequency.
	//	- CTRL_REG2_G = 0x00: HPF set to normal mode, cutoff frequency
	//		set to 7.2 Hz (depends on ODR).
	//	- CTRL_REG3_G = 0x88: Interrupt enabled on INT_G (set to push-pull and
	//		active high). Data-ready output enabled on DRDY_G.
	//	- CTRL_REG4_G = 0x00: Continuous update mode. Data LSB stored in lower
	//		address. Scale set to 245 DPS. SPI mode set to 4-wire.
	//	- CTRL_REG5_G = 0x00: FIFO disabled. HPF disabled.
	void initGyro();
	
	// initAccel() -- Sets up the accelerometer to begin reading.
	// This function steps through all accelerometer related control registers.
	// Upon exit these registers will be set as:
	//	- CTRL_REG0_XM = 0x00: FIFO disabled. HPF bypassed. Normal mode.
	//	- CTRL_REG1_XM = 0x57: 100 Hz data rate. Continuous update.
	//		all axes enabled.
	//	- CTRL_REG2_XM = 0x00:  2g scale. 773 Hz anti-alias filter BW.
	//	- CTRL_REG3_XM = 0x04: Accel data ready signal on INT1_XM pin.
	void initAccel();
	
	// initMag() -- Sets up the magnetometer to begin reading.
	// This function steps through all magnetometer-related control registers.
	// Upon exit these registers will be set as:
	//	- CTRL_REG4_XM = 0x04: Mag data ready signal on INT2_XM pin.
	//	- CTRL_REG5_XM = 0x14: 100 Hz update rate. Low resolution. Interrupt
	//		requests don't latch. Temperature sensor disabled.
	//	- CTRL_REG6_XM = 0x00:  2 Gs scale.
	//	- CTRL_REG7_XM = 0x00: Continuous conversion mode. Normal HPF mode.
	//	- INT_CTRL_REG_M = 0x09: Interrupt active-high. Enable interrupts.
	void initMag();
```
and **PASTE** it in this same file right above the word "_protected_" (making the functions part of the _public_ / exposed functions)

4. Save the file (overwriting the exising one)

After this Re-compile your Arduino project and make sure to _uncomment_ these two statements
```
#define ENABLE_SENSOR
#define USE_PROTECTED_MEMBERS
```

wide-area-devices, 2020-08-25:
You are a star, I'll try that today, does this mean it's possible to get close to halting the IMU or is it to get it down to ~1ma, which is still an obstacle for the little battery, all I'm after is temperature. Having taken off the RTC also I've picked up I2C from wires and using an HTU21D sensor instead. At present it wakes once a minute and sends via wifi then sleeps for a minute, on a not full battery I got 247 mins running, which makes me feel my once an hour and sleep for a few days might be reachable, if the IMU only goes down to 1ma it'll have a huge impact

btw "air rework" makes my £23 YORKING 858D from amazon sound more amazing than it is :))))

genotix, 2020-08-25:
> You are a star, I'll try that today, does this mean it's possible to get close to halting the IMU or is it to get it down to ~1ma, which is still an obstacle for the little battery, all I'm after is [...]

I believe it is.
My watch is now running for 15 hours (occasional use) and the battery is at 56%.
Without this change it would make it to about 7 to 8 hours.

It really was the IMU killing the battery but I do not have a scope attached to see the exact usage.

genotix, 2020-08-25:
I'll try to add this to the **SparkFunLSM9DS1.cpp** file to get the temperature

```
int16_t LSM9DS1::getTemp()
{
	readTemp();
	return temperature;
}
```
Don't expect it to be very precise.

boogie, 2020-08-25:
@genotix Wow! Is it working without modifying the hardware? Respect for figuring this out.

genotix, 2020-08-25:
> @genotix Wow! Is it working without modifying the hardware? Respect for figuring this out.

Well I surely do get a way better usage this way.
Still looking if this can be tweaked somewhat more.

genotix, 2020-08-25:
Sadly I just broke my screen of the LiliGo. :(
My next step would have been to make the update OTA by long pressing the wake button.

Opened a github page with my current developments:
https://github.com/genotix/LilyGO-T-Wristband

genotix, 2020-08-29:
Did some measurements:

00:10 => 89%
09:33 => 63%
19:30 => 48%
24:19 => 47%
34:19 => 25%
46:30 => Empty 

So with this latest sketch it would function for about 40 hours I think.

As soon as I get the new watch I'll continue tuning. 
(broke the screen of this one)
I also intend to dismantle the old watch to use it on a more sensitive Amp meter.

wide-area-devices, 2020-08-29:
Not sure why, maybe it's my multimeter but everything I try gets the circuit to be 1.77-2.09ma at best, this would mean a sleep time of 40ish hours at best

By taking the RTC and IMU off and putting on a different I2C tempsensor (controlled by gpio pin) I'm down to more like 0.2ma

I'm now looking towards the D4pico modules to see where I can get to with that and might if I'm unlucky have to prototype my own design that fits in the same case

Lilygo should have thought more in the design stage for sure

genotix, 2020-08-30:
> Not sure why, maybe it's my multimeter but everything I try gets the circuit to be 1.77-2.09ma at best, this would mean a sleep time of 40ish hours at best
>
>
>
> By taking the RTC and IMU off and [...]

Well that seems to be spot-on compared to my latest tests.

Being honest I am not sure what else we could do other than start desoldering components.

I agree with you on the fact that LiliGo could have done more to make the device more Low energy. Even with these on-board sensors. Surely a v2 could fix that and I would surely buy that.

biccius, 2021-02-08:
Going a little deeper into the initialization of the LSMD9DS1 I noticed that there is a call to the function init() that initialize the device with the default settings of Sparkfun that obviously do not take into account the sleep mode.

[image]

So I think that any settings made before are overwritten by this call

I suggest to comment it out

Personally I was able to go well over 24 hours of battery life (I'm using a 300mah) using appropriately deep sleep, cpuclock at 10mhz with jumps to 80mhz to make ble connections when necessary and accellerometer settings to low power

v44r, 2021-07-03:
I'm getting 0.43mA of current consumption on the t-wristband (MPU9250) with >4V on the battery right now... That's good enough for a few days on the 40mA battery (I have a vibration module so I cannot put the larger batteries on it).

The key is putting the MPU9250 to sleep (both accel/gyro AND magnetometer). Otherwise you get ~3.75mA. 

This gets it down to 0.7mA:

	uint8_t b = _bus->readByte(MPU9250_ADDRESS, PWR_MGMT_1);
	_bus->writeByte(MPU9250_ADDRESS, PWR_MGMT_1, b | (1 << 6));

And this down to 0.43mA:

	_bus->writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x00);

That's after normal usage (i.e., turn on wifi, accel/gyro/mag, tft, etc. and then turn them off and go to deep sleep leaving RTC alarm wakeup and button wakeup on):

	esp_sleep_enable_ext0_wakeup((gpio_num_t) RTC_INT_PIN, LOW); // wake by RTC 
	esp_sleep_enable_ext1_wakeup(GPIO_SEL_33, ESP_EXT1_WAKEUP_ANY_HIGH); // wake by touch
	esp_deep_sleep_start();

ErickRBTK, 2021-08-03:
Les hace falta ver mas box.
se puede conseguir mas autonomia:
1 usando deepsleep cuando no se ocupa el ESP32
2 apaga la pantalla despues de 5 segundos
3 solo enciendes el wifi cuando lo usas
4 apaga el IMU si no lo usas
5 modificar el brillo pantalla segun la hora
6 bajar la velocidad del CPU

Tengo un MPU9250 con vibrador a 40mA. He conseguido unos 3 dias como reloj y boton wifi para mis proyectos. accel para cambiar horientacion, gyro off, mag off. en la foto ya lleva unas 12hrs desde que lo desconecte. trabajando en:
- brujula
- wom
- podometro
- alarma

https://1drv.ms/u/s!ArJwl0aLcuHssArITlP9b0-pt5qz

hpsaturn, 2021-08-06:
@ErickRBTK the thread is in English. But anyway, all here we already reached 3 days of duration, and I think that all here has implemented the same improvements.