Xinyuan-LilyGO/T-Wristband #5: Could not write

a9b, 2020-02-02:
I want help
Compilation succeeded, but could not write

This is Error message
```
Arduino：1.8.10 (Mac OS X), ボード："ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, Verbose"
tool.py v2.6
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_____....._____....._____....._____....._____....._____.....____スケッチの書き込み中にエラーが発生しました
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```

"ESP32 Dev Module" is selected in the procedure of the image below,
Port is "cu.SLAB_USBtoUAR"
https://github.com/Xinyuan-LilyGO/LilyGO-T-Wristband#3-precautions

lewisxhe, 2020-02-13:
Make sure your FPC terminals are not connected incorrectly

lewisxhe, 2020-06-28:
If there are still problems, please reopen the issue

closed 2020-06-28 by lewisxhe