Xinyuan-LilyGO/T-Wristband #6: IMU_INT_PIN 38 or 39?

qgar12, 2020-02-02:
according to shematics IMU_INT_PIN =39
according to code & markdown IMU_INT_PIN =38

whats the right value?

Thx

TioRuben, 2020-02-02:
In the case of the unit I've is PIN 39

TioRuben, 2020-02-09:
FYI I've successfully set up wake on motion feature by sending this to MPU
```c++
  IMU.writeBit(MPU9250_ADDRESS, PWR_MGMT_1, 5, false);
  IMU.writeBit(MPU9250_ADDRESS, PWR_MGMT_1, 6, false);
  IMU.writeBit(MPU9250_ADDRESS, PWR_MGMT_1, 4, false);
  IMU.writeByte(MPU9250_ADDRESS, PWR_MGMT_2, 0x07);
  IMU.writeByte(MPU9250_ADDRESS, ACCEL_CONFIG2, 0x09);
  IMU.writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x40);
  IMU.writeBit(MPU9250_ADDRESS, MOT_DETECT_CTRL, 7, true);
  IMU.writeBit(MPU9250_ADDRESS, MOT_DETECT_CTRL, 6, true);
  IMU.writeByte(MPU9250_ADDRESS, WOM_THR, 170);
  IMU.writeBit(MPU9250_ADDRESS, INT_PIN_CFG, 5, 0);
  IMU.writeByte(MPU9250_ADDRESS, INT_STATUS, 0);
  IMU.writeByte(MPU9250_ADDRESS, LP_ACCEL_ODR, 0x03);
  IMU.writeBit(MPU9250_ADDRESS, PWR_MGMT_1, 5, true);
```
[Invensense Datasheet](invensense.com/wp-content/uploads/2015/02/PS-MPU-9250A-01-v1.1.pdf) Pages 30 and 31 **"6.1 Wake-on-Motion Interrupt"**

And, setting up the EXT1 Wakeup
```c++
  pinMode(39, GPIO_MODE_INPUT);
  esp_sleep_enable_ext1_wakeup(GPIO_SEL_39, ESP_EXT1_WAKEUP_ANY_HIGH);
  esp_deep_sleep_disable_rom_logging();
  esp_deep_sleep_start();
```

Now I can move the arm towards my face to see the hour without touching any button.

lewisxhe, 2020-06-28:
If there are still problems, please reopen the issue

closed 2020-06-28 by lewisxhe