Xinyuan-LilyGO/T-Wristband #9: OTA Hotspot not accessible

MauiKano, 2020-03-01:
I can flash the precompiled firmware T-Wristband-20200104 which brings me the WiFi manager 
hotspot T-Wristband and I can configure WiFi. All works. Then I compile the latest Arduino source found in Xinyuan-LilyGo with the development branch of the WiFi manager. I can load the resulting firmware to my Wristband and it goes to the hotspot and shows the respective screen on the wristband but the WiFi is not announced. I can not find it when searching Wifi on my computer of phone. Any ideas ?

sysdl132, 2020-03-01:
Do you have a firewall?Or your library is no problem?You can try the master branch.
(Anyone can fix it or have a good idea?)

MauiKano, 2020-03-01:
I have a firewall but that is not the issue. the wristband hotspot works with the precompiled firmware. 
I use the Wifi manager library by cloning that way:
git clone -b development https://github.com/tzapu/WiFiManager.git

sysdl132, 2020-03-01:
[image]
Do you tried master branch?

sysdl132, 2020-03-01:
[image]

MauiKano, 2020-03-01:
using the master branch I get a compile error:
Used: ...../sketchbook/hardware/espressif/esp32/libraries/WiFi
.........sketchbook/libraries/WiFiManager/WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory

sysdl132, 2020-03-01:
Make sure you installed ESP8266 wifi library.
Maybe you can see README.md in WifiManager.
[image]

sysdl132, 2020-03-01:
You  can check them:
[image]
(I am not make sure)

MauiKano, 2020-03-01:
Doubt that these are needed.

How does the source code of the firmware T-Wristband-20200104  look like ? I am particularly interested how setupWiFi looks like. The current code is this: Is there a difference ?

void setupWiFi()
{
#ifdef ARDUINO_OTA_UPDATE
    WiFiManager wifiManager;
      
    //set callback that gets called when connecting to previous WiFi fails, and enters Access Point mode
    wifiManager.setAPCallback(configModeCallback);
    wifiManager.setBreakAfterConfig(true);          // Without this saveConfigCallback does not get fired
    wifiManager.autoConnect("T-Wristband");
#endif
}

is

MauiKano, 2020-03-04:
I found it :)
My hardware installation for ESP32 in the Arduino IDE was outdated. I installed fresh from here:
https://github.com/espressif/arduino-esp32
That solved the issue.

closed 2020-03-04 by MauiKano

sysdl132, 2020-03-04:
Don't forget to update your outdated installation next time:D

MauiKano, 2020-03-05:
well. I had HW configured in arduono IDE via the json URL in preferences. That should give you auto update. However, there are various ESP32 HW branches out there. It would be good if you explicitly mention in the README the ESP32 git URL that is needed.

But i have a second issue now: A WiFi configuration is not stores persistently. If my wristband goes to sleep and rewoke again i have to configure wifi again:(