Xinyuan-LilyGO/T-Wristband #11: Button is mechanically unstable

donatoaz, 2020-03-07:
Has anyone else experienced that the button never really works? I've tried it with several different debounce libraries and even my own code, and it never really works as expected, as if there was some really strange mechanical problem.

Has anyone tried to change the button for something else, or has any suggestion on how to troubleshoot it?

PolyMeilex, 2020-03-07:
For me it worked only when I pressed it really hard. I suspect it was same problem as yours, sadly I didn't find any solution for this. And it makes whole product literally unusable for me

donatoaz, 2020-03-07:
Yep, literally unusable for me as well, just lost half a saturday.

cpetito, 2020-03-12:
I also had an initial issue with the touch sensor.  I found that the sticky metal mesh on the underside of the cover needed to be moved a bit over so it made better contact with the spring contact on the PC board.

I suggest putting this code in your loop logic so you can see how the sensor is responding:

    if (digitalRead(TP_PIN_PIN) == HIGH) {
      digitalWrite(LED_PIN, HIGH);
    }
    else {
      digitalWrite(LED_PIN, LOW);
    }

With this code you can try touching the sensor pin with the cover off and see how that compares with touching the indent with the cover on.

I also noticed that when I put the cover on or put the module in the wrist strap, the sensor stays on for a few seconds.  This could be due to the TTP223-BA6's auto calibration, but I have not been able to find any information regarding the calibration algorithm.

jithurbide, 2020-04-27:
Hello,

I had a lot of issue with mine. Watch waking up without touching it. Button not working as expected. To solve it, I remove the conductive pad and move it from horizontal to vertical. I discovered that the connection from pad to pcb was not correct. May be it’s your problem. Try to repositioning the conductive pad.