Xinyuan-LilyGO/T-Wristband #15: TECH SUPPORT IS NOT RESPONDING

boogie, 2020-06-13:
Wrote to them about the battery issues of this product, and no response.
**DO NOT BUY THIS PRODUCT!!!**

boogie, 2020-06-15:
They have answered me. They say it is for development purposes, if you want better battery life, I have to upgrade the battery. An upgraded battery won't help a lot.

If you don't want to use this product, but want to learn development, it might work, but I think there are better products for that.

lewisxhe, 2020-06-18:
Your question has nothing to do with the code, please contact LilyGo if you have a product question

closed 2020-06-18 by lewisxhe

boogie, 2020-06-18:
@lewisxhe Can you please share **the code** that improves the battery life? I got no answer from you how I can reach the 400uA consumption in sleep mode. It IS a technical issue.