Xinyuan-LilyGO/T-Wristband #22: Impossible programming?

inoffice, 2020-10-27:
Hi and good day

Serial port COM7
Connecting........_____....._____....._____....._____....._____....._____.....____Ha ocurrido un error mientras se enviaba el sketch

I try hold button of program PCB and same!
How i can programming?
BEST REGARDS

genotix, 2020-10-27:
1. Lower your baud rate and try again.
2. If it doesn’t work, goto 1

Met vriendelijke groet / Kind regards,

Eric Oud Ammerveld
Phone:
+31 10 495 19 55
Mobile:
+31 6 55 86 25 44
E-mail:
eric@genotix.nl

inoffice, 2020-10-28:
Hi and good day
Great Resolve!! thanks

genotix, 2020-10-28:
Good luck!

Met vriendelijke groet / Kind regards,

Eric Oud Ammerveld
Phone:
+31 10 495 19 55
Mobile:
+31 6 55 86 25 44
E-mail:
eric@genotix.nl

closed 2022-09-05 by liangyingy