Xinyuan-LilyGO/T-Wristband #24: How NRF52832 Module is programming?

furkanisikay, 2020-12-01:
Hello,

I bought a nrf52832 model ttgo t wristband. I did not program with mpu9250 example or lms91d example. it did not work.

Arduino ide give says ;
Sketch uses 719322 bytes (54%) of program storage space. Maximum is 1310720 bytes.
Global variables use 40352 bytes (12%) of dynamic memory, leaving 287328 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6
Serial port COM4
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

Also, I tried lower bandwith but it did not work.

How I program this module? Which library can I use for programming this module?

genotix, 2020-12-01:
Lower your communication speed to 115200

Met vriendelijke groet / Kind regards,

Eric Oud Ammerveld
Phone:
+31 10 495 19 55
Mobile:
+31 6 55 86 25 44
E-mail:
eric@genotix.nl

furkanisikay, 2020-12-03:
It did not work.

genotix, 2020-12-03:
Then you might have either a faulty programmer, connection or chip

furkanisikay, 2020-12-04:
I tried with 8 another programmer.I tried with 3 another cables and ports. In addition, I programmed another mpu 9250 models with programmer that I used before. Finally, my idea is "we have to use another sdk for this models" or "I have product that is crashed".

genotix, 2020-12-04:
Hmm sounds like it is crashed?

Met vriendelijke groet / Kind regards,

Eric Oud Ammerveld
Phone:
+31 10 495 19 55
Mobile:
+31 6 55 86 25 44
E-mail:
eric@genotix.nl

fgmn, 2022-03-11:
I have the same problem, have you solved it?