Xinyuan-LilyGO/T-Wristband #25: Yaw value problem

fabioetec, 2020-12-06:
I would like to develop a compass.

So I modified MPU9250 example creating a sprite with an arrow which I rotate with the angle returned by IMU.yaw.

If I correctly understand IMU.yaw should give me the angle in degrees from north to the current orientation.

The problem is IMU.yaw return me completly random values, for example without moving the device and setting `#define SerialDebugMPU9250 true` on sensor.h I get:

```
ax = 42.85 ay = -302.12 az = -1009.95 mg
gx = -0.11 gy = 0.30 gz = 0.05 deg/s
mx = -69 my = -468 mz = -464 mG
-0.10-0.100.83-0.54
Yaw, Pitch, Roll: 
-185.50, -15.81, -114.78
ax = 42.79 ay = -307.19 az = -1008.85 mg
gx = -0.21 gy = 0.23 gz = 0.03 deg/s
mx = -91 my = -483 mz = -478 mG
-0.200.410.870.18
Yaw, Pitch, Roll: 
124.09, -29.93, 170.05
ax = 44.19 ay = -302.19 az = -1005.49 mg
gx = -0.10 gy = 0.36 gz = -0.05 deg/s
mx = -69 my = -472 mz = -471 mG
-0.10-0.090.83-0.54
Yaw, Pitch, Roll: 
-186.66, -15.36, -114.67
ax = 42.48 ay = -305.11 az = -1008.97 mg
gx = -0.12 gy = 0.14 gz = 0.04 deg/s
mx = -85 my = -474 mz = -497 mG
-0.210.420.860.19
Yaw, Pitch, Roll: 
122.11, -30.88, 170.19
ax = 41.20 ay = -301.39 az = -1010.62 mg
gx = -0.21 gy = 0.43 gz = -0.03 deg/s
mx = -96 my = -481 mz = -483 mG
-0.11-0.090.84-0.53
Yaw, Pitch, Roll: 
-186.20, -16.45, -115.61
ax = 42.60 ay = -303.28 az = -1006.04 mg
gx = -0.02 gy = 0.25 gz = 0.08 deg/s
mx = -85 my = -474 mz = -470 mG
-0.190.390.880.19
Yaw, Pitch, Roll: 
126.80, -28.78, 168.31
ax = 45.04 ay = -302.37 az = -1007.51 mg
gx = -0.20 gy = 0.28 gz = 0.06 deg/s
mx = -80 my = -486 mz = -489 mG
-0.11-0.110.83-0.53
Yaw, Pitch, Roll: 
-184.33, -17.75, -115.75
ax = 45.10 ay = -302.80 az = -1007.93 mg
gx = -0.14 gy = 0.25 gz = -0.02 deg/s
mx = -82 my = -481 mz = -476 mG
-0.200.390.880.18
Yaw, Pitch, Roll: 
125.85, -29.06, 169.68
ax = 44.25 ay = -305.48 az = -1007.39 mg
gx = -0.18 gy = 0.30 gz = 0.03 deg/s
mx = -93 my = -481 mz = -476 mG
-0.11-0.110.83-0.53
Yaw, Pitch, Roll: 
-184.56, -17.01, -115.13
ax = 41.38 ay = -305.11 az = -1006.71 mg
gx = -0.21 gy = 0.29 gz = 0.01 deg/s
mx = -78 my = -481 mz = -463 mG
-0.200.400.880.17
Yaw, Pitch, Roll: 
124.80, -28.98, 170.71
ax = 42.24 ay = -301.03 az = -1010.74 mg
gx = -0.23 gy = 0.54 gz = 0.03 deg/s
mx = -78 my = -481 mz = -490 mG
-0.10-0.080.83-0.55
Yaw, Pitch, Roll: 
-186.81, -14.53, -113.13
ax = 43.82 ay = -303.59 az = -1009.28 mg
gx = -0.13 gy = 0.31 gz = -0.04 deg/s
mx = -85 my = -477 mz = -483 mG
-0.220.430.860.18
Yaw, Pitch, Roll: 
119.90, -31.47, 172.18
ax = 42.36 ay = -302.61 az = -1008.12 mg
gx = -0.16 gy = 0.31 gz = 0.02 deg/s
mx = -80 my = -490 mz = -464 mG
-0.10-0.070.84-0.53
Yaw, Pitch, Roll: 
-187.59, -14.18, -115.61
ax = 42.79 ay = -301.39 az = -1007.75 mg
gx = -0.15 gy = 0.13 gz = -0.02 deg/s
mx = -80 my = -472 mz = -478 mG
-0.200.410.870.19
Yaw, Pitch, Roll: 
123.40, -30.35, 168.94
ax = 42.60 ay = -301.21 az = -1006.04 mg
gx = -0.18 gy = 0.20 gz = 0.18 deg/s
mx = -82 my = -481 mz = -487 mG
-0.12-0.100.84-0.53
Yaw, Pitch, Roll: 
-185.76, -17.32, -116.07
```

moving from positive to negative values without any sense.

biccius, 2020-12-07:
Do you have a code snippet to simulate it?

fabioetec, 2020-12-07:
> Do you have a code snippet to simulate it?

Sure! On line 10 of this file https://github.com/Xinyuan-LilyGO/LilyGo-T-Wristband/blob/master/examples/T-Wristband-MPU9250/sensor.h change:

`#define SerialDebugMPU9250 0`

with

`#define SerialDebugMPU9250 true`

then open serial monitor and, with the button on the device, go to the page where you see accelerometer values (green text on black background). On the serial monitor, you will see the output printed by this code (lines 185-192 same sensor.h file, you can also remove comment on line with header "Yaw, Pitch, Roll: ")

 ```
   if (SerialDebugMPU9250) {
      // Serial.println("Yaw, Pitch, Roll: ");
      Serial.print(IMU.yaw, 2);
      Serial.print(", ");
      Serial.print(IMU.pitch, 2);
      Serial.print(", ");
      Serial.println(IMU.roll, 2);
    }
```

Without moving the device I should get almost the same number for yaw, pitch and roll, instead I get differents values at every loop.

For any other question don't hesitate to ask!

biccius, 2021-02-08:
Sorry, but i don't have MPU9250
With LSM9DS1 i'm able to get valid pitch and roll values using this snippet of code

`void LilyGo::getCurrentOrientation()
{
	this->update_imu();

	x = imu.calcAccel(imu.ax);
	y = imu.calcAccel(imu.ay);
	z = imu.calcAccel(imu.az);

	pitch = atan2(x, sqrt(y * y) + (z * z));
	roll = atan2(y, sqrt(x * x) + (z * z));
	pitch *= 180.0 / M_PI;
	roll *= 180.0 / M_PI;
`

Hope it can help